fossasia/pslab-app #1542: Oscilloscope crashes application

vincentmkes, 2019-02-08:
**Actual Behaviour**

Oscilloscope causes application to crash

**Expected Behaviour**

Oscilloscope should actually launch

**Steps to reproduce it**

Issue on Nexus 7 with Android 6.0.1

1.Connect the pslab card
2.Open the Application
3.Launch Oscilloscope
Application crashes

cweitat, 2019-02-12:
@vincentmkes hello, did you sent the crash log?

vincentmkes, 2019-02-12:
Yes I did - if there is anything else I can do let me know.
I also had the same issue with Android 5.0 but can’t upgrade any further than that

closed 2019-04-29 by cweitat

vincentmakes, 2019-04-29:
Is it closed because it is solved ?

cweitat, 2019-05-03:
No other reports of it crashing and can't reproduce. Do try with other phones and if it still happens please reopen this issue.

vincentmakes, 2019-05-03:
Other reviews on the Google Playstore are stating similar issue so that's disappointing you are ignoring it. I don't have any other Android hardware. How can I get a refund ? Thanks

cweitat, 2019-05-08:
I saw that there was an update of the app version in the store. please try that out.