fossasia/pslab-app #1557: No Back Arrow on action bar of LuxMeter and BaroMeter

neel1998, 2019-02-20:
**Actual Behaviour**

There is now Back arrow on the action bar of Lux Meter and Baro Meter, which inconsistent with other Meters' activity layout

**Expected Behaviour**

There should be a back arrow

**Steps to reproduce it**

Open Lux Meter or Baro Meter

**Screenshots of the issue**

[image]
[image]

**Would you like to work on the issue?**

Yes

ho-dor, 2019-02-20:
We don't have to add back button as @cloudypadmal has mentioned in #1543

neel1998, 2019-02-20:
Okay...I didn't see that issue. I will close this issue

closed 2019-02-20 by neel1998