fossasia/pslab-app #1576: Display a toast when power source values values reach minimum or maximum

ho-dor, 2019-03-04:
**Actual Behaviour**

No way to find out the max and min value for PV1 , 2 and 3.

**Expected Behaviour**

Display a toast if up button is clicked when value is maximum or if down button is clicked if value is minimum.

**Steps to reproduce it**

Open Power source and click PV1 after it's value is -5.00 V
**LogCat for the issue**

Not applicable

https://drive.google.com/open?id=1cUcVUu2F9h6AIftDk6rnOhSQZmmfX56F

**Would you like to work on the issue?**

YES

CloudyPadmal, 2019-03-04:
When the values are at minimum or a maximum, the knob is turned to far left or right respectively. Is that functionality not working?

ho-dor, 2019-03-04:
It's working .

ho-dor, 2019-03-05:
@CloudyPadmal should i work on this or is this unnecessary ?

CloudyPadmal, 2019-03-06:
For now let's work on some other issues as this seems unnecessary. :+1:

closed 2019-03-06 by CloudyPadmal