fossasia/pslab-app #1594: Home Screen message to tell about the pin layout

riyasingh1004, 2019-03-06:
**Actual Behaviour**

There is no message to tell a new user where to see the pin layout

**Expected Behaviour**

A Toast message to tell them where to look for pin layout

**Would you like to work on the issue?**

yes

riyasingh1004, 2019-03-06:
I think this will be better than implementing a start message for guiding new users

neel1998, 2019-03-06:
@riyasingh1004  adding so many toast at every activity decrease the User experience, Since the pin layout option is in the action bar menu , there is no need to add such toast messages

riyasingh1004, 2019-03-06:
I thought this Toast will replace the earlier one. This would. be more useful because in the beginning even I had to search for the pin layout. It would guide them where to look

riyasingh1004, 2019-03-06:
Definitely all the Toast messages will not be implemented. I am just suggesting all the possibilities.@neel1998

mariobehling, 2019-03-29:
Need to think about user communication, but this seems a bit too much at the moment.

closed 2019-03-29 by mariobehling