fossasia/pslab-app #1598: Feature Request: Altitude Meter

neel1998, 2019-03-07:
**Actual Behaviour**
Currently no altitude meter

**Expected Behaviour**
Since there is barometer, we can calculate the altitude for those devices which supports barometer
**Would you like to work on the issue?**
Yes

@harsh-2711  @CloudyPadmal  @cweitat kindly share your view on this feature

mariobehling, 2019-03-29:
Nice idea!

neel1998, 2019-04-09:
@mariobehling  @CloudyPadmal  @harsh-2711 @abhinavraj23  @cweitat should i add the altitude information in the barometer activity or should there be a separate activity only for altitude meter. If we are adding altitude meter to the barometer activity, how should we display the graph of the altitude meter? Kindly shed some light on this matter

neel1998, 2019-04-17:
@mariobehling  @CloudyPadmal  kindly share your views on how to add altitude meter. I have given two options in the previous comment.

neel1998, 2019-05-11:
@CloudyPadmal  @abhinavraj23  please share some suggestion for this feature. Should it be included in barometer itself or there should be separate activity like other sensors.

ho-dor, 2019-06-03:
@neel1998 what about this ?

PR #1924 (2019-08-15): feat: feature to see altitude from pressure added in Barometer
Fixes #1598 

**Changes**: feature to see altitude from pressure added in Barometer

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kb [... cut]
closed 2019-08-15 by CloudyPadmal