fossasia/pslab-app #1606: Add option in FAQ page where user can ask questions other than once already availbale

neel1998, 2019-03-07:
**Actual Behaviour**
Currently user can read answers to only some questions built in
**Expected Behaviour**
There should be option so that user can ask questions other than those already available

**Would you like to work on the issue?**
Yes. 

@harsh-2711  @cweitat  @CloudyPadmal  I need your view on solving this issue. I think there are 2 options : 
1) create a google form similar to feedback form, and add link to that in FAQ page 
2) Create a small form in the FAQ page itself where user can feel in email and question and send.

Kindly suggest if any better option comes to your minds :)

ho-dor, 2019-03-07:
@neel1998 I think there is the feedback & bugs link for this .

neel1998, 2019-03-07:
@ho-dor  feedback and bugs are completely different from FAQs. There should be an separate option for FAQ certainly

harsh-2711, 2019-03-08:
@neel1998 I think making a Google form is much convenient way else the page will be cluttered.  But I think this feature isn't needed now rather working on instruments is needed

ho-dor, 2019-05-04:
@neel1998 what is the status of this ? I think it can be closed as @harsh-2711 mentioned.

closed 2019-05-04 by cweitat