fossasia/pslab-app #1612: Accelerometer's "Swipe to view guide" not functioning

neel1998, 2019-03-08:
**Actual Behaviour**

Accelerometer's "Swipe to view guide" not functioning 
**Expected Behaviour**

Guide should open on swiping up
**Steps to reproduce it**

Open accelerometer and try to swipe up

**Would you like to work on the issue?**
yes

ho-dor, 2019-03-08:
@neel1998 #1541 fixes this.

neel1998, 2019-03-08:
@ho-dor  Okay. Thanks. Can you fix the issues with that PR. It has been pending since long

closed 2019-03-08 by neel1998

ho-dor, 2019-03-08:
@neel1998 no changes have been requested as of now . Since , it is a big PR , the review is taking long and @abhinavraj23 and @CloudyPadmal are reviewing it.

neel1998, 2019-03-08:
@ho-dor it is still failing codacy review..You can try fixing that first

ho-dor, 2019-03-08:
@neel1998  the codacy errors involve "long method names" and "use explicit scoping" due to @BindView used for scoping and it has been followed in other files as well and as for the long method names , the name is in accordance with used in other activities. I have solved the unused declarations and imports errors caused due to codacy earlier.