fossasia/pslab-app #1616: Autoscan button squeezed in Sensors Activity

harshdevl, 2019-03-09:
**Actual Behaviour**

Button width is squeezed while reformatting the project (120dp -> 20dp)

**Expected Behaviour**

Button width should be 120dp in Sensors Activity

**Steps to reproduce it**

Go to Sensors Activity -> Observe

**LogCat for the issue**

N/A

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

ho-dor, 2019-03-09:
One PR #1541 is open for this which solves the error . It is under review .

harshdevl, 2019-03-09:
I can't see any related issue in your PR. The issue that you are solving is #1465 which handles crash in Accelerometer and Power Source instrument.

PR #1618 (2019-03-21): fix: Solved issues in Sensor activity
Fixes #1616 #1617 

**Changes**: Added missing shadow view and changed the width of the buttons in dimensions file

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd [... cut]
closed 2019-03-21 by mariobehling