fossasia/pslab-app #1622: Missing back button

chanmol1999, 2019-03-18:
**Actual Behaviour**

Their is no back button in instruments in case of Barometer and Lux meter.

**Expected Behaviour**

Their should be a navigating back button so that same interface is followed for all instruments.

**Steps to reproduce it**

change in activity or fragment.

**LogCat for the issue**

**Screenshots of the issue**

[image]
[image]

**Would you like to work on the issue?**
Yes

PalAditya, 2019-03-22:
Can I Work on this issue?

chanmol1999, 2019-03-22:
No I am working on it and will soon do a pr .You can search for another issue:)

neel1998, 2019-03-22:
@chanmol1999  we don't need a back button here as @CloudyPadmal  mentioned in #1543 Kindly check closed issues before raising a new one. You can close this one

ho-dor, 2019-03-29:
Already a navigation menu has been added to these instruments .

Smit2010, 2019-04-03:
Anybody working on this?

neel1998, 2019-04-03:
@Smit2010  I guess we won't need a back button since consistent action bar is to be added to all the devices.

Smit2010, 2019-04-03:
@neel1998 I guess @mariobehling suggested to implement the same back arrow to all the instruments rather than implementing menu.

Smit2010, 2019-04-08:
@CloudyPadmal Is this required now? I would like to take this up.

PR #1683 (2019-05-13): Fix # 1622, 1653 back button added to required instruments
Fixes #1622 #1653

**Changes**: I have added back button to action bar of PSLab Sensor activity. Hence all the instruments that inherits this activity will have this back button

**Screenshot/s for the changes**: 
[image]
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I  [... cut]
closed 2019-05-13 by mariobehling