fossasia/pslab-app #1625: Barometer BM180 I2C produce random values

gdampf, 2019-03-19:
**Actual Behaviour**

conected BMP180 based on GY-68 is detected in Sensors, but not in Barometer (shows not connected).
trying to measure produces very unstable data (Temperature show random values between 10 and 400 degree centigrade). But you can see, if you filter just the lowest values, that this raises, if you warm the sensor.

**Expected Behaviour**

More or less stable temperature, if not warming or cooling the sensor

**Steps to reproduce it**

Connect a BMP180 based on GY-68 (4-pin-connector: GND, VIN, SCL, SDA) to corresponding pins as described and auto-detect sensors, than choose BMP180 external and start measurement.

**Would you like to work on the issue?**

No -  too messy project.

Screenshot:
[image]

kartikeysaran, 2021-06-04:
Should i work on this ? @CloudyPadmal @mariobehling

CloudyPadmal, 2021-06-04:
Hello @kartikeysaran , without a device it would be challenging. But you can give it a try. 👍🏼

kartikeysaran, 2021-06-05:
> Hello @kartikeysaran , without a device it would be challenging. But you can give it a try. 👍🏼

Sure ! :)

marcnause, 2024-08-05:
Will be fixed in https://github.com/fossasia/pslab-android/pull/2527

marcnause, 2024-08-07:
Fixed via https://github.com/fossasia/pslab-android/pull/2527

closed 2024-08-07 by marcnause