fossasia/pslab-app #1628: Oscilloscope: Clicking options on bottom area opens top red bar

mariobehling, 2019-03-24:
The oscilloscope doesn't show the top bar of the phone. However when the user clicks options in the white bottom bar area the top bar shows up. Expected: top bar should not show up.

[image]

ho-dor, 2019-03-24:
can I work on this @mariobehling ?

neel1998, 2019-03-24:
@mariobehling  I guess this issue was solved in #1562 . This issue doesn't occur on my app. Kindly check once .
[image]

neel1998, 2019-04-03:
@mariobehling  this can be closed

neel1998, 2019-04-11:
@CloudyPadmal  @mariobehling  @harsh-2711 @abhinavraj23  I think this issue is solved by #1556  and can be closed.

ho-dor, 2019-06-09:
@mariobehling this can be solved . It is resolved by #1556

CloudyPadmal, 2019-06-10:
This is fixed in #1556

closed 2019-06-10 by CloudyPadmal