fossasia/pslab-app #1629: New instrument : Thermometer

ho-dor, 2019-03-25:
**Actual Behaviour**

There is no device to calculate temperature in the app.

**Expected Behaviour**

Using the inbuilt android ambient temperature sensor , create a thermometer.

**Steps to reproduce it**

Not applicable

**LogCat for the issue**

Not Applicable

**Screenshots of the issue**

Not applicable

**Would you like to work on the issue?**

Yes .

neel1998, 2019-03-25:
@ho-dor  this might not be very much useful since very few smartphones out there have external temperature sensor. All phones have internal temperature sensor for the main chip, but external temperature sensor is not provided with most of the smart phones. @harsh-2711  @mariobehling  kindly share your views on this issue

ho-dor, 2019-03-25:
This is same for gyroscope or the magnetic sensor which isn't present in Samsung J8 and I think some other devices as well

neel1998, 2019-03-25:
@ho-dor  looking at the statistics of the the sensors in the amar phones currently in market . Very very small percent provide ambient temperature sensor. However number of smart phones having gyroscope is very large. Still you can implement this feature. There nothing to lose having more feature :)

ho-dor, 2019-03-29:
@mariobehling please give your views on this new instrument .

CloudyPadmal, 2019-03-29:
Hello @ho-dor , Thermometer would be a good idea. We are not focusing on sensors in Android phone but the sensors that can be connected with PSLab externally. I think [LM35](www.ti.com/lit/ds/symlink/lm35.pdf) would be a better candidate for this. Could you please provide us with some mockups? Thanks. If a device has a thermometer inbuild, we can let user select which one to chose.

ho-dor, 2019-03-29:
@CloudyPadmal sure . I will let you know soon.

ho-dor, 2019-05-22:
@CloudyPadmal can you take a look at SHT21 sensor already used in the app , can I use that in place of LM35 ?

CloudyPadmal, 2019-05-22:
Yea that would be ok. But we can let user choose which sensor to use and give an option to switch between sensors. You can start with anything you like.

neel1998, 2019-05-22:
@ho-dor  Are you planning on adding a new instrument activity or making changes in the sensors activity?

ho-dor, 2019-05-22:
@neel1998 for now , a new instrument activity.

neel1998, 2019-05-22:
@ho-dor  I think it would be better to have it in sensors activity. Because i still think most of the mobiles don't have external temperature sensor. And those wit the PSLab board can use the sensor from the sensor activity. @CloudyPadmal  kindly share your opinion. I think we should have separate instrument activities for those sensors which are more likely to be present in more android devices

ho-dor, 2019-05-22:
@neel1998 but we are going to give option to the user to choose the inbuilt device sensor or through the PSLab device .

neel1998, 2019-05-22:
@ho-dor  alright :). Let me know if you need any help

PR #1737 (2019-05-28): Added Thermometer UI
Fixes #1629 

**Changes**: Added UI for thermometer

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines or extra spaces
- [x] I [... cut]
closed 2019-05-28 by mariobehling