fossasia/pslab-app #1630: Automatic collapse other question when some other question is expanded in FAQ

ho-dor, 2019-03-25:
**Actual Behaviour**

When one question is expanded , it remains expanded even after another question is expanded . If some questions stay expanded , the bottom questions go out of the screen.

**Expected Behaviour**

If some other question is expanded then the previously expanded question should  collapse.

**Steps to reproduce it**

Go to FAQ and click on all questions.

**LogCat for the issue**

Not Applicable.

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes.

neel1998, 2019-03-25:
I don't think this required at this moment. Since there very few questions in FAQ. @harsh-2711  @mariobehling  kindly share your views

ho-dor, 2019-03-25:
@neel1998 as you can see in the screenshot , we will have to scroll to see the bottom question and as questions will increase, this will improve the User experience as scrolling is not considered a good experience . Thank you

neel1998, 2019-03-25:
@ho-dor  alright.

mariobehling, 2019-03-30:
Actually, this is just a personal preference and other people like to read longer info. So, I don't think it is a priority at the moment. Let's focus on high priority issues: https://github.com/fossasia/pslab-android/issues?q=is%3Aissue+is%3Aopen+label%3A%22Priority%3A+High%22

closed 2019-03-30 by mariobehling