fossasia/pslab-app #1632: Add functionality to continuously record data in compass

neel1998, 2019-03-25:
**Actual Behaviour**
Currently record function in compass stores instantaneous value 

**Expected Behaviour**
Compass should record data until stop button is pressed , like other instruments

**LogCat for the issue**
N/A

**Screenshots of the issue**
N/A

**Would you like to work on the issue?**
Yes

mariobehling, 2019-03-29:
Please provide an example of an existing app or mockups. So, we can better understand how you want to implement this feature.

neel1998, 2019-04-09:
@mariobehling  currently the compass logs instantaneous values. I am suggesting implemet loggin function in compass just like barometer and lux meter, so user can start/stop the logging and can see the result for a time period not just for the given instance. Kindly share your thought. @CloudyPadmal @harsh-2711 @abhinavraj23  @cweitat please take a look at this issue. Thanks

PR #1688 (2019-05-12): Fix #1632 #1635 #1644 Continuous logging added to compass
Fixes #1632 #1635  #1644 

**Changes**: 
I have made following changes

- **Inherited** compass activity from **PSLabSensor activity** for consistency throughout the code
- Added functionality to **continuously log and play** data in compass instrment
- The logged data of compass was **not visible in Logged Data activity** previously. I have fixed it
- in the CSV generated for the log of compass, **selected axis and Location** is added, which was previuosly missing.

**Screenshot/s for the changes**: 
[image]
[image]
[image]

**Checklis [... cut]
closed 2019-05-12 by CloudyPadmal