fossasia/pslab-app #1634: Show message to user when gyroscope not present in device

ho-dor, 2019-03-25:
***Actual Behaviour**

Compass is not working on Samsung J8 device as it does not have any gyroscope .

**Expected Behaviour**
A toast or a dialog should appear to tell user that there is no gyroscope in this device so compass will not work.

**Steps to reproduce it**

On a Samsung J8 device , open compass

**LogCat for the issue**

Not Applicable

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

neel1998, 2019-03-25:
@ho-dor  i think compass uses built in magneto meter and not gyroscope. Kindly check once . @harsh-2711  since you worked on implementation of compass kindly check this once

ho-dor, 2019-03-25:
Either way , we should give a message to user when any sensor isn't present . @neel1998  I can change gyroscope to any sensor

neel1998, 2019-03-25:
@ho-dor i am not saying issue is wrong, i am just saying make sure you check availability of correct sensor. Cheers!

PR #1637 (2019-05-11): Added Toast message when sensor not present in Compass
Fixes #1634 

**Changes**: A toast message appears if the device dos not contain magnetic sensor

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd> [... cut]
closed 2019-05-11 by CloudyPadmal