fossasia/pslab-app #1636: New Instrument : GyroScope

neel1998, 2019-03-25:
**Actual Behaviour**
Currently the app has accelerometer and compass, though both may seem similar to gyroscope, functionality of gyroscope is quite different. Gyroscope provides absolute rotation of device, where accelerpmeter give changes in orientation , whereas compass uses magnetic sensors. Since almost every smart phone has gyroscope built in, this would be a good feature to include

**Would you like to work on the issue?**
Yes

neel1998, 2019-03-29:
@mariobehling i think this would be a nice feature to include. If you agree i would start working on it. Thanks

mariobehling, 2019-03-29:
Thanks, please provide a mockup or screenshot of an existing app, that has this feature. Please include Guide, top menu bar with burger menu etc.

PR #1672 (2019-05-13): Fix #1636 New instrument gyroscope added
Fixes #1636

**Changes**: 

- I have added a new feature gyroscope which uses devices built in sensor to measure rate of rotation
- I have implemented recording functionality for gyroscope as well
- Layout of the device is kept same as accelerometer
- Implementation is similar to Barometer and Lux meter for consistency  

**Screenshot/s for the changes**: 

[image]
[image]
[image]
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `co [... cut]
closed 2019-05-13 by mariobehling