fossasia/pslab-app #1638: Decimal digits in default values of lux meter inconsistent

ho-dor, 2019-03-26:
**Actual Behaviour**

There are unequal number of decimal digits in default values of lux meter

**Expected Behaviour**

Consistent number of decimal digits should be there

**Steps to reproduce it**

Open Lux meter

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes.

Smit2010, 2019-04-08:
@ho-dor Are you still working on this? If not, I would like to take this up.

ho-dor, 2019-04-08:
Ya sure . You can work on this . Thanks .

Regards,
Kunal Rai(BT16CSE058),
NATIONAL INSTITUTE OF TECHNOLOGY , UTTARAKHAND,
+91-7895780897

PR #1680 (2019-04-09): Decimals in Luxmeter made consistent
Fixes #1638 

**Changes**:All the readings have been to rounded off to 2 decimal place.

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does [... cut]
closed 2019-04-09 by abhinavraj23