fossasia/pslab-app #1640: Change title bar of about us

chanmol1999, 2019-03-26:
**Actual Behaviour**
Currently all the items in navigation drawer have their title names in their activity except ,about  page for which title of activity is PSlab.

**Expected Behaviour**

Also just like other activities it should be about instead of Ps Lab.

**Steps to reproduce it**
Open About page.

**LogCat for the issue**

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**
yes

VishalBhosale5, 2019-03-27:
May I take up this issue please

chanmol1999, 2019-03-27:
No I am working on this and will soon make pr.

VishalBhosale5, 2019-03-27:
I got it, can I fix this.

PR #1652 (2019-03-29): Change title  bar of About Us
Fixes #1640 
**Changes**:strings.xml

**Screenshot/s for the changes**:
[image]
**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [ x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [ ] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [ ] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x ] My code does not contain any extra lines or extra spaces
- [ ] I have  [... cut]
closed 2019-03-29 by mariobehling