fossasia/pslab-app #1642: Lux Meter Recording should not be allowed if there is no sensor in device

ho-dor, 2019-03-26:
**Actual Behaviour**

Even if there is no sensor , recording is allowed . Wastes memory.

**Expected Behaviour**

If no sensor , don't allow recording

**Steps to reproduce it**

If device doesn't have lux sensor then open lux meter and click recording.

**LogCat for the issue**

Not Applicable

**Screenshots of the issue**

Not Applicable

**Would you like to work on the issue?**

Yes

Smit2010, 2019-04-08:
@ho-dor Are you still working on this? If not, I would like to work on this.

ho-dor, 2019-04-08:
I think I have created a PR for this . Thanks .

Regards,
Kunal Rai(BT16CSE058),
NATIONAL INSTITUTE OF TECHNOLOGY , UTTARAKHAND,
+91-7895780897

ho-dor, 2019-05-15:
Closing this as this has been merged

closed 2019-05-15 by ho-dor