fossasia/pslab-app #1650: Record option for oscilloscope, logic analyzer and wave generator

neel1998, 2019-03-27:
**Actual Behaviour**
Currently there is no option to record data for these instruments

**Expected Behaviour**
Add option to record.

**Would you like to work on the issue?**
Yes. @harsh-2711 @CloudyPadmal  @cweitat  @mariobehling  can you please guide and suggest me whether this feature should be added or not

ho-dor, 2019-04-09:
@neel1998 are you working on this ? There are many high priority issues pending . I can help if you don't mind in this or ahy other high priority issue . Thank You.

neel1998, 2019-04-09:
@ho-dor  I will be working on this issue. But still there has not been any confirmation from the mentors regarding the format in which the data needs to be logged. Once that is finalise i will file a PR. Thanks

neel1998, 2019-04-19:
@CloudyPadmal  @cweitat  @harsh-2711 @abhinavraj23  kindly provide suggestions for this issue. So that i can start working on it

neel1998, 2019-05-11:
@CloudyPadmal  @abhinavraj23  kindly provide some guidance for this issue so that i can start working on it. This might take time

CloudyPadmal, 2019-05-11:
What kind of guidance you need?

neel1998, 2019-06-17:
@CloudyPadmal  the wave generator save config is done and working fine. I wanted to know how are we planning to save data for oscilloscope, like what will be the fields, also when users play backs a saved log, what should be exactly show?

CloudyPadmal, 2019-06-17:
Data we need to log will be
- Readings related to each channel enabled by user
- Time
- Location
- Timebase/Amplitude settings per each reading

When user plays it back, we need to display the waveform at the same interval as they were recorded at the same amplitudes.

neel1998, 2019-06-17:
Okay. But what if while recording user changes the tab, as in goes to DataAnalysis tab or fourier tab, will we accordingly change the tab?...What we can do here is have a separate screen for playing the logged data of oscilloscope, which just has graph , even if during the recording user has changed to fourier transform, we can accordingly show fourier output for that period. Does that sound okay ?

CloudyPadmal, 2019-06-17:
We keep on recording the raw data even if the use changes tabs because we will still have the feed. We don't need to record fourier data separately. We can derive that from oscilloscope readings as we are doing now. In the playback, let's give user the option to view either waveform or frequency graphs. We'll have to make some UI modifications for that.

neel1998, 2019-06-17:
Alright. I think i should make a separate screen to show the recorded data, since there is no sufficient space in current oscilloscope UI, further we need to add play/pause/stop and fourier transform toggle buttons. Does that sound good?

CloudyPadmal, 2019-06-17:
We might be able to use the same UI by disabling components that are not needed and updating the components and their values. Say a user starts recording channel 1. Then he selects channel 2 and increase timebase. We can reuse the components to show user when he has selected channel 2 and the value in timebase slider. We can ditch the XY plot button and use that space and tab for some playback features

neel1998, 2019-06-17:
But opening a same activity when user plays the data and then changing the UI programmatically will make a lots of redundant code. Won't be sensible to have a separate activity for playback with similar UI as the actual oscilloscope activity with necessary additions and exclusions

CloudyPadmal, 2019-06-17:
Redoing a whole new UI not redundant?

neel1998, 2019-06-17:
I think redundant was an incorrect word i used :laughing: .. Alright I will try adding recording feature in the same activity.

CloudyPadmal, 2019-06-17:
Yes, we need to implement features quickly and efficiently. Making new UIs and new classes would take days and at the end of GSoC it will be just the oscilloscope. Let's speed things up.

neel1998, 2019-06-17:
Got it. On it.

PR #1802 (2019-06-20): Fix #1650 logging added to oscilloscope
Fixes #1650

**Changes**: logging added to oscilloscope

**Screenshot/s for the changes**: 
[image]

[image]

[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contai [... cut]
closed 2019-06-20 by mariobehling