fossasia/pslab-app #1653: Make action bar of instruments activities consistent.

neel1998, 2019-03-27:
**Actual Behaviour**
Currently all the actionbars are different. Some have hamburger, some have back arrow, some have nothing at all.

**Expected Behaviour**
Make the actionbars consistent

**Would you like to work on the issue?**
Yes .
 @CloudyPadmal  you earlier mentioned to add hamburger to barometer and lux meter. If that is the design suited for the app .. i think hamburger should be added to multimeter, power source, accelerometer and compass as well.. kindly suggest what needs to be done. Because current design looks inconsistent.  @harsh-2711  @cweitat  kindly share your views

Smit2010, 2019-03-27:
I am already working on this issue.

neel1998, 2019-03-27:
I see you have solved the issue for barometer and lux meter. But there has not been confirmation from maintainers regarding whether hamburger should be added to other instruments or not

neel1998, 2019-03-27:
@Smit2010  I think we should wait until maintainers confirma the required design

Smit2010, 2019-03-28:
@neel1998 Before taking up this issue, I have already talked to @CloudyPadmal about this and he suggested to first send a PR for these two instrument and then for the others. I have already implemented for all the instruments in my code base. I am waiting for the mentors to accept my PR.

ho-dor, 2019-04-11:
This is really confusing. Hamburger icon in some and back button in some instruments . What do you think @neel1998 @abhinavraj23 @CloudyPadmal ?

neel1998, 2019-04-11:
@ho-dor  I think @mariobehling  alrady mentioned earlier to have back button. I have made the PR adding back button. I guess after that PR is merged every instrument will have a back button.

closed 2019-05-23 by neel1998