fossasia/pslab-app #1657: Add Show guide option to multimeter menu

neel1998, 2019-03-28:
**Actual Behaviour**
Currently there is no option for show guide in multimeter menu. Which is inconsistent with menu of barometer and lux meter and other instruments

**Expected Behaviour**
There should be a show guide menu

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

mariobehling, 2019-03-29:
Thanks for this issue. We would like to become more efficient. Instead of making a single small issue for each instrument, please help to resolve this and add the Guide to any missing instrument. Thanks.

Smit2010, 2019-04-08:
@neel1998 Are you still working on this? If not, I would like to work on this.

neel1998, 2019-04-08:
@Smit2010  sorry. I am working on this. Since i busy completing my proposal I could not make a PR. will file a PR soon.

PR #1684 (2019-05-03): Fix #1657 show guide option added
Fixes #1657 
**Changes**: Every instrument except multimeter had show guide option, so i have added that option to multimeter menu

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>AL [... cut]
commit 83e2fd26 (2019-05-03): Merge pull request #1684 from neel1998/show_guide_option_all

Fix #1657 show guide option added
closed 2019-05-03 by cweitat