fossasia/pslab-app #1662: No Location data stored in CSV even though "include location data" option is checked

neel1998, 2019-03-29:
**Actual Behaviour**

Even if user has selected to add location data in .csv logs, the latitude and longitude columns have zero values in generated csv

**Expected Behaviour**
The current values of latitude and longitude should be included in csv when "include location data" option is checked.

**Steps to reproduce it**

Go to any instruments. (eg. Lux menter) select "include location data" option in configuration menu. Record some data. Check the generated .csv file

**Would you like to work on the issue?**
yes

CloudyPadmal, 2019-03-29:
Hi, could you please check if this can be implemented with open street maps? We are not using Google Maps FYI.

neel1998, 2019-03-29:
@CloudyPadmal  I will surely look into it. Will update you if I make ant progress. Thanks

PR #1677 (2019-05-15): Fix #1662 location coordinates issue solved
Fixes #1662 

**Changes**: removed `locationReady` boolean from `GPSLogger` class because it was never set to true because `onLocationChanged` callback is not called always

**Screenshot/s for the changes**:
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in ever [... cut]
closed 2019-05-15 by CloudyPadmal