fossasia/pslab-app #1664: Power source guide not opening by sliding up

ho-dor, 2019-03-29:
**Actual Behaviour**

Guide of compass is not opening from the bottom of the screen by sliding up. Guide does get opened by '?' Sign on taskbar but it should also open by sliding up like in the case of oscilloscope.

**Expected Behaviour**

On sliding up , guide should open from bottom of screen.

**Steps to reproduce it**

Open compass and slide up from bottom of the screen

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

neel1998, 2019-03-29:
@ho-dor  i think this is issue with all the activities which have scroll bars. Same as compass. Kindly resolve the issue for all. As @mariobehling suggested . Don't open separate issue for each instrument. Fix this and all similar when you fix the compass one. Also change compass to power source in the text of this issue. Cheers!

ho-dor, 2019-03-29:
Thanks.  I will take a look.

closed 2019-05-04 by ho-dor