fossasia/pslab-app #1668: Crash : App crashes when About Us clicked in Nav menu

ho-dor, 2019-03-31:
**Actual Behaviour**

App is crashing when About Us in Nav menu clicked

**Expected Behaviour**

App should not crash

**Steps to reproduce it**

Go to nav menu and click on About Us

**LogCat for the issue**

Process: io.pslab, PID: 18344
    java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.SubMenu.getItem(int)' on a null object reference
        at io.pslab.activity.MainActivity.selectNavMenu(MainActivity.java:228)
        at io.pslab.activity.MainActivity.loadHomeFragment(MainActivity.java:153)
        at io.pslab.activity.MainActivity.access$300(MainActivity.java:52)
        at io.pslab.activity.MainActivity$2.onNavigationItemSelected(MainActivity.java:297)
        at android.support.design.widget.NavigationView$1.onMenuItemSelected(NavigationView.java:170)
        at android.support.v7.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:840)
        at android.support.v7.view.menu.SubMenuBuilder.dispatchMenuItemSelected(SubMenuBuilder.java:90)
        at android.support.v7.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
        at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:991)
        at android.support.design.internal.NavigationMenuPresenter$1.onClick(NavigationMenuPresenter.java:352)
        at android.view.View.performClick(View.java:6909)
        at android.view.View$PerformClick.run(View.java:26199)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6944)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**

Yes.

neel1998, 2019-04-02:
@ho-dor  this issue can be closed now

neel1998, 2019-04-11:
@ho-dor  please close this issue

closed 2019-04-11 by ho-dor