fossasia/pslab-app #1673: No option to open logged data in compass

ho-dor, 2019-04-01:
**Actual Behaviour**

No option to open logged data in compass

**Expected Behaviour**

An option to view/open logged data should be there.

**Steps to reproduce it**

Open compass and record some data

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**
Yes.

neel1998, 2019-04-01:
I am working on adding logged data option to all the instruments. I will cover compass as well

neel1998, 2019-04-01:
#1645 #1635  will solve this one as well... I think this issue can be closed

ho-dor, 2019-04-01:
Ok . Thanks .

closed 2019-04-01 by ho-dor