fossasia/pslab-app #1674: New Feature : Add functionality to save graph views in gallery

neel1998, 2019-04-01:
**Actual Behaviour**

Currently there is no functionality to save graphs

**Expected Behaviour**
User should be able to save the graph of some instance in gallery using a button click

**Steps to reproduce it**
N/A

**LogCat for the issue**
N/A

**Screenshots of the issue**
N/A

**Would you like to work on the issue?**
yes

neel1998, 2019-04-01:
@mariobehling  @CloudyPadmal  @cweitat  @harsh-2711 I need some suggestions for this feature. I am thinking of adding a button next to record button on the action bar for the instrument. upon clicking would capture the current state of the graph on the screen and store it as jpeg. Kindly share your views and suggestions

ho-dor, 2019-04-11:
@neel1998 since we can already record the graph readings , what purpose will this function serve ? Thanks.

neel1998, 2019-04-11:
@ho-dor  this feature will save the image of the graph in the device. So you can share it if you want. Unlike the csv it will be visual representation which much more comprehensive . Also this functionality has been added as a TODO list in the code as well.

ho-dor, 2019-04-11:
@neel1998 when I think about the use that you have mentioned , I think the person who will receive this graph image will also need at what time in the experiment was the graph captured . What do you think ? Maybe a timer can be added when recording is started and hence we will receive the exact time during the experiment this was captured .

neel1998, 2019-04-11:
Kindly explain the timer part again. The X-axis of the graph is anyways time(s). So we can easily understand the time that is passed since the experiment was started from the image of the graph.

ho-dor, 2019-04-11:
What I am trying to say is that we can extend this functionality beyond the instruments with graph like Barometer reading for exact instant , adding a timer there and then share it might be useful.

neel1998, 2019-04-11:
Most of the instruments have graphs and they are the only ones which requires logging in my opinion. Only compass and Multimeter dont have graphs. We can think something about them, once this basic functionality is working

PR #1696 (2019-05-10): Fix #1674 save graph function implemeted
Fixes #1674 

**Changes**: When save graph button is clicked on the data logger activity, if the log file alrady exist in the local storage then only the .jpg image of current graph is stored in the local storage, if user has deleted the log from local storage then the log file created and .jpg file is stored. 

So even if user deletes the log from local storage it can again be generated using the app, which was not possible earlier.

This feature is currently added to Lux meter and Baro meter
**Checklist**: [Please tick following check boxes with `[x]` [... cut]
closed 2019-05-10 by CloudyPadmal