fossasia/pslab-app #1682: Indication of current position in MainActivity shown wrong

Smit2010, 2019-04-08:
**Actual Behaviour**
When user selects "Share App" from the navigation drawer while in Device, FAQs or About us of the segment represented in drawer, the indication always indicates Instruments irrespective of the current position.

**Expected Behaviour**
Indicator should show the current location in drawer.

**Steps to reproduce it**
Open any of the options in menu (Device, FAQs, About us) then click "Share App" from the drawer >> click cancel and open the drawer and notice the indicator (Highlighted menu) and also the app bar title.

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes, I would like to work on this issue.

neel1998, 2019-04-08:
@Smit2010  kindly give comprehensible title to the issue. Current title doesn't quite explain the real issue

neel1998, 2019-04-08:
@ho-dor  can you check whether this issue occurs after your changes in the PR #1676

neel1998, 2019-04-08:
@Smit2010 please check once that your repo is updated after the PR #1671  was merged. May be this issue could have been fixed in that PR. If not you can work on it

Smit2010, 2019-04-09:
@neel1998 After thePR #1671, the issue still persists. So I am continuing working on it.

PR #1685 (2019-04-10): Indicator of current position in Navigation Menu improved
Fixes #1682 

**Changes**: Indicator of current position in navigation drawer had bugs while clicking "Share App" option in the menu. Bug is solved.

**Screenshot/s for the changes**: 
[image]
**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file inc [... cut]
closed 2019-04-10 by abhinavraj23