fossasia/pslab-app #1687: Add the long press functionality to other instruments like in Power Source

ho-dor, 2019-04-11:
**Actual Behaviour**

No message appears on long press in other instruments like in Power Source "Show Guide".

**Expected Behaviour**

"Show Guide message should appear on long press of "Show Guide" item.

**Steps to reproduce it**

Go to Power Source and long press "?" symbol

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

Not applicable

**Would you like to work on the issue?**

Yes

neel1998, 2019-04-11:
@harsh-2711  @abhinavraj23  @CloudyPadmal  why do we have this functionality. Anyways "?" is self explanatory regarding the use of that button and also single click is more prompt response than a long press. User is like to click the "?" button before long pressing, and that would anyway open the guide.This is still my point of view.

ho-dor, 2019-04-11:
@neel1998 I agree. Either we can remove this from all instruments or add it to all . Consistency was my motive when I opened this issue.

neel1998, 2019-04-11:
@ho-dor  I understand your point. Let's wait for mentors opinion regarding this.

PR #1692 (2019-05-11): Added Long Press functionality to remaining instruments
Fixes #1687 

**Changes**: Changes made in 3 instruments which didn't have long press functionality like other instruments  This brings consistency in all instruments.

**Screenshot/s for the changes**:
[image]

[image]
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [ ] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have  [... cut]
closed 2019-05-11 by CloudyPadmal