fossasia/pslab-app #1689: Support OLED 128x64 Displays

CloudyPadmal, 2019-04-14:
**Actual Behaviour**

Android app doesn't have an option to use OLED display

**Expected Behaviour**

Android app needs to support OLED displays. When the app opens, it needs to show an initiation screen.

**Steps to reproduce it**

> Not available

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Open to new contributors

neel1998, 2019-04-14:
@CloudyPadmal  can you please elaborate on the issue, and what is the desired output. I would relly love to contribute

ho-dor, 2019-04-14:
@CloudyPadmal I am interested . Please guide how to go about it . Thanks.

CloudyPadmal, 2019-04-15:
Excellent. I'll give you more details and specifications.

Smit2010, 2019-04-15:
I would also like to work on this. Please guide me how get started for this.
@CloudyPadmal

neel1998, 2019-04-19:
@CloudyPadmal  any updates on this issue?

ho-dor, 2019-07-08:
@CloudyPadmal can you tell a bit more about this? Are we supposed to provide this functionality with help of an external sensor? I was looking in the list of i2c sensors and found 2 which are used for oled SSD1305 and SSD1306? [https://learn.adafruit.com/i2c-addresses/the-list](url)

neel1998, 2019-07-08:
@ho-dor i already have the display and am working on it.

CloudyPadmal, 2019-07-08:
We need I2C to work for this to function properly. For now, we can let user display any string or shape on the oled screen

neel1998, 2019-07-08:
For basic functionality we will use SPI , right?. I tried running the example code for spi on my arduino with oled

CloudyPadmal, 2019-07-08:
We'll use the I2C display. That's the common one. Less wires.

neel1998, 2019-07-08:
Oh. I have one with 6 pins. So that won't work?

CloudyPadmal, 2019-07-08:
This is I2C. It'll work