fossasia/pslab-app #1690: Playback data from SD Card

CloudyPadmal, 2019-04-14:
**Actual Behaviour**

PSLab v6 has SD card feature to log recording from different instruments. These recording will be in raw values with no visualization.

**Expected Behaviour**

Android app needs to support a feature to let user play back these recorded data by importing them from SD card and display in a similar manner to logged data in current app.

**Steps to reproduce it**

> Not available

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

New contributors can take up this issue and discuss with me how to integrate with the firmware

ho-dor, 2019-04-14:
@CloudyPadmal an you please explain how to go about this . I am interested .

neel1998, 2019-04-14:
@CloudyPadmal  i am also interested to work on this. Kindly guide how to go about it.

Smit2010, 2019-04-15:
@CloudyPadmal I am keen on working on this. Please guide to get started.

PR #1691 (2019-05-11): WIP: Playback data from SD card feature added
Fixes #1690 

**Changes**: 

I have added an option in data logger activity where user can import the log stored as CSV on the SD card for the instrument selected by the user. 

**Currently I have implemented this feature for Barometer and Lux meter. Once the mentors approves these changes I will add the other instruments as well**

**Screenshot/s for the changes**: 

Following are the steps to import log in the application

**step 1**

Create a CSV file on the SD card or local storage with proper headers and required fields as per selected instrum [... cut]
closed 2019-05-11 by CloudyPadmal