fossasia/pslab-app #1694: Saved file of logged data not opening

Smit2010, 2019-04-20:
**Actual Behaviour**
When user tries to save the logged data and open that particular file saved in local storage from the smackbar, the user is redirected to the same fragment of Logged Data.

**Expected Behaviour**
Opening the saved location should access the file manager and open the PSLab folder where that particular file is saved.

**Steps to reproduce it**
Open Logged Data >> open any present data >> save the data from top right >> click on "open" in smackbar that appears below.

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes.

neel1998, 2019-04-20:
@Smit2010 the "save" button on the snack bar is not meant to save the log. it is for saving graph. IF you check the code of "PSLabSensor" activity this menu calls saveGraph() function in corresponding fragment, which is currently empty. I have already opened an issue #1674  to  implement this and have started working on that. I think this issue can be closed. Thanks

abhinavraj23, 2019-04-20:
@neel1998 I think this is a completely different issue, it just states that the file is not getting opened from the app, the issue which you referred to is different than this and moreover is pending from 20 days. You can continue your work with this issue @Smit2010

neel1998, 2019-04-20:
@abhinavraj23  sorry for the confusion. But since the button that this issue refers to calls the **save graph function** i was confused. Which still I am. **What should the save button exactly do?**. And regarding the pending issue, I was solving other issues and was a bit busy with my ends. Will make a PR as soon as possible. (Also my intention from previous comment was certainly not to restrict other contributors to make new contribution, i was genuinely confused that's why raised the comment.)

abhinavraj23, 2019-04-20:
@neel1998 I thought this issue is regarding opening the file manager through the app, but I guess there are some other issues as well, so the best thing to do in these kinds of cases is to communicate with among the developers, that is the reason we have IRC channels where you guys can talk and discuss related to the app, if you want you can also directly talk with each other privately as well in Gitter.  So the best would be to discuss this issue among yourselves and come up with a possible solution rather than discussing more here!

neel1998, 2019-05-10:
@Smit2010  are you still working on this?. If not I would like to take it up

CloudyPadmal, 2019-05-10:
I think this is an invalid issue. When user opens the log, with the current implementation it opens a list of all logged data specific to that instrument. I don't think seeing a bunch of raw values (lot of numbers) would make any sense. But with the current view it's showing, user can play back the data visually and see location details as well. So I'm closing this issue.

closed 2019-05-10 by CloudyPadmal

neel1998, 2019-05-10:
@CloudyPadmal  I think there is some misunderstanding. This issue states that, when user clicks on "save graph" button while seeing a recorded log, the "open" button on the snackbar takes user to the logged data activity, instead it should take user to local file storage where the log is stored. So if user want to take some actions with that log file (like sharing it), user can easily do that

neel1998, 2019-05-11:
@CloudyPadmal  any update on this issue. I think it is a valid one

CloudyPadmal, 2019-05-11:
I'm not sure if it's a valid one. Leave it for now.