fossasia/pslab-app #1705: Implement save graph function for accelerometer

neel1998, 2019-05-11:
**Actual Behaviour**
Currently save graph function is not implemented for accelerometer

**Expected Behaviour**
Simpleng the save graph function like barometer and lux meter

**Would you like to work on the issue?**
Yes

CloudyPadmal, 2019-05-12:
May be we can implement a prototype in PSLabSensor class so that it will be available everywhere?

neel1998, 2019-05-12:
I will check if that can be done.

neel1998, 2019-05-12:
@CloudyPadmal in my opinion adding a prototype function in PSLabSensor for saveGraph() function would make the implementation a bit complex. Because all the instruments have different CSV formats. also to save the graph as .jpg we need to access linear layout of each instrument, doing all these through a common function would require to add many getters in the respective instrument activities. Kindly share your views.

CloudyPadmal, 2019-05-12:
Ok, let's dig deep and see in the coming days. For now you can continue with this issue.

PR #1724 (2019-05-22): save graph function added to compass, gyroscope and accelerometer
Fixes #1705 #1709 #1712

**Changes**:Implemented saveGraph function to gyroscope, compass and gyroscope

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code doe [... cut]
closed 2019-05-22 by mariobehling