fossasia/pslab-app #1716: Improvement: Gyroscope instrument is inconsistent

Smit2010, 2019-05-14:
**Actual Behaviour**
Major functionalities are working fine but user guide, configurations and logged data is inconsistent.

**Expected Behaviour**
Alignment in configuration should be consistent with luxmeter and barometer, clicking map option in logged data should work, adding text in user guide and setting appropriate message and limit in Update time and High limit in configurations.

**Steps to reproduce it**
Open app -> open Gyroscope and observe its functionalities.

**Screenshots of the issue**
[image] [image] [image]

**Would you like to work on the issue?**
Yes, I would like to work on this issue.

neel1998, 2019-05-14:
@Smit2010  let me know if you find any trouble going through the implementation of gyroscope :)

Smit2010, 2019-05-14:
Yeah sure. @neel1998

PR #1718 (2019-05-17): Improvements in Gyroscope instrument
Fixes #1716 

**Changes**:
Gyroscope instrument made consistent by fixing bugs. Alignment in configuration made consistent with luxmeter and barometer, clicking map option in logged data working, added text in user guide and setting appropriate message and limit in Update time and High limit in configurations. 

**Screenshot/s for the changes**:
[image] [image] [image] [image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` withou [... cut]
closed 2019-05-17 by CloudyPadmal