fossasia/pslab-app #1719: Servo motor control option

CloudyPadmal, 2019-05-15:
**Actual Behaviour**

PSLab device can be used to control a robot arm set with 4 servo motors. Currently it is done by writing a Python script and there is no option in PSLab Android app to do so. This feature is requested by Marco

**Expected Behaviour**

Implement a tool giving an option to control servo motors using Android app.

**Steps to reproduce it**

> Not available

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Add this to GSoC timeline. Developer who takes over this issue needs to understand how servo motors work and how to use PWM pins of PSLab device to control them.

neel1998, 2019-05-15:
@CloudyPadmal  i have worked with servo motors earlier. Kindly elaborate a little bit further on this so that I can start working on this

adityastic, 2019-05-15:
It was my minor to control servo's using android, haha what a coincidence. @neel1998 good luck

CloudyPadmal, 2019-05-15:
We have a robot arm that moves with 4 servo motors. They are controlled with a code similar to
```python
from PSL import sciencelab
Robot = sciencelab.connect()
Robot.servo4(30,30,30,30)
```

Idea is to make it more user friendly and give user the ability to change these `30` values to different values so the arm will move differently. You get the idea. It'll be just calling a function but have to make it catchy

adityastic, 2019-05-15:
@CloudyPadmal that would be great

neel1998, 2019-05-15:
@CloudyPadmal  that's great

adityastic, 2019-05-15:
@CloudyPadmal can you show us the real look of the robotic arm, I would love to create something exactly like it inside the app

CloudyPadmal, 2019-05-15:
> @CloudyPadmal can you show us the real look of the robotic arm, I would love to create something exactly like it inside the app

This would be hard. Something like [image]

neel1998, 2019-05-15:
@adityastic  instead of creating a arm like layout in the app , i was thinking of having a co troller for each four motor, which would make user able to move the servo as interested. It would look more neat

CloudyPadmal, 2019-05-15:
Just some sliders would be good without a complicated UI

neel1998, 2019-05-15:
@CloudyPadmal  i agree with you

adityastic, 2019-05-15:
@CloudyPadmal got it

adityastic, 2019-05-15:
> @adityastic instead of creating a arm like layout in the app , i was thinking of having a co troller for each four motor, which would make user able to move the servo as interested. It would look more neat

What about 4 dials? that would be neat for an initial interface

neel1998, 2019-05-15:
@adityastic  i was also thinking something like that. Let's both create some mock ups and then decide which layout to opt

adityastic, 2019-05-15:
> @adityastic i was also thinking something like that. Let's both create some mock ups and then decide which layout to opt

Sounds Great, Lets talk about it on the public channel on gitter

neel1998, 2019-05-16:
@CloudyPadmal  i have made 2 mock ups for this functionality. Kindly share your views so that i can start working on it.
[image]

[image]

CloudyPadmal, 2019-05-17:
First one looks more comfortable to slide :thinking:

mariobehling, 2019-05-17:
As the numbers symbolize degrees in a circle, please also use a circle for the UI. Ensure that the degrees e.g. 90° have the correct position in the circle.

mariobehling, 2019-05-17:
Furthermore give the option that uses can enter values as numbers if they click/tap on the number area.

neel1998, 2019-05-17:
@mariobehling I will start working on it as per your suggestion

adityastic, 2019-05-17:
@neel1998 looks goood

PR #1732 (2019-05-24): WIP #1719 layout to control servo of robotic arm created
Fixes #1719 

**Changes**: Created a layout to control servo

**Screenshot/s for the changes**: 
[image]

[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code do [... cut]
closed 2019-05-24 by mariobehling