fossasia/pslab-app #1723: Logged Data not showing graph in Gyroscope

ho-dor, 2019-05-15:
**Actual Behaviour**

When logged data is opened , it does not show the graph in devices with sensor absent.

**Expected Behaviour**

The graph should show either zero like when recorded or disable recording when sensor not present.

**Steps to reproduce it**

Open gyroscope with device not having the sensor and record some data and open it.

**LogCat for the issue**

Not Applicable

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

ho-dor, 2019-05-15:
@CloudyPadmal  should we disable recording when sensor not present or add the graph ?

neel1998, 2019-05-15:
@ho-dor kindly change the title of the issue to include that this issue occurs only when sensor is not present.

ho-dor, 2019-06-10:
> @CloudyPadmal should we disable recording when sensor not present or add the graph ?

@CloudyPadmal @neel1998 what are your thoughts on this ?

CloudyPadmal, 2019-06-10:
@ho-dor , disable the graph and give some visual output to user about the situation

PR #1785 (2019-06-11): Visual message when no data fetched in gyroscope
Fixes #1723

**Changes**: Graphs removed and a visual message to user when no data fetched

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [ ] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [ [... cut]
closed 2019-06-11 by CloudyPadmal