fossasia/pslab-app #1728: Outofmemory crash in PSLabPinLayoutFragment

CloudyPadmal, 2019-05-17:
**Actual Behaviour**

Google Play crash reports indicate that there is a crash happening when a user tries pinmap multiple times. Probably some caches not getting clear.

**Expected Behaviour**

App should not crash and memory needs to be utilized

**Steps to reproduce it**

[Download and view steps recorded from Google Play tests](https://github.com/fossasia/pslab-android/files/3189457/OutOfMemoryGooglePlay.zip)

**LogCat for the issue**

```
05-04 07:30:07.963: E/AndroidRuntime(11854): FATAL EXCEPTION: ControllerMessenger
05-04 07:30:07.963: E/AndroidRuntime(11854): java.lang.OutOfMemoryError
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.graphics.BitmapFactory.nativeDecodeAsset(Native Method)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.graphics.BitmapFactory.decodeStream(BitmapFactory.java:596)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.graphics.BitmapFactory.decodeResourceStream(BitmapFactory.java:444)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.graphics.drawable.Drawable.createFromResourceStream(Drawable.java:832)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.content.res.Resources.loadDrawable(Resources.java:2988)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.content.res.Resources.getDrawable(Resources.java:1558)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v7.widget.VectorEnabledTintResources.superGetDrawable(VectorEnabledTintResources.java:75)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v7.widget.AppCompatDrawableManager.onDrawableLoadedFromResources(AppCompatDrawableManager.java:432)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v7.widget.VectorEnabledTintResources.getDrawable(VectorEnabledTintResources.java:68)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.content.res.ResourcesCompat.getDrawable(ResourcesCompat.java:84)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at io.pslab.fragment.PSLabPinLayoutFragment.onCreateView(PSLabPinLayoutFragment.java:60)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.Fragment.performCreateView(Fragment.java:2439)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1460)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1784)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1852)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.BackStackRecord.executeOps(BackStackRecord.java:802)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.FragmentManagerImpl.executeOps(FragmentManager.java:2625)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2411)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2366)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2273)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.support.v4.app.FragmentManagerImpl$1.run(FragmentManager.java:733)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.os.Handler.handleCallback(Handler.java:730)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at android.os.Handler.dispatchMessage(Handler.java:92)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at androidx.test.espresso.base.Interrogator.a(Interrogator.java:19)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at androidx.test.espresso.base.UiControllerImpl.a(UiControllerImpl.java:166)
05-04 07:30:07.963: E/AndroidRuntime(11854): 	at androidx.test.espresso.base.UiControllerImpl.a(UiControllerImpl.java:158)
```

**Screenshots of the issue**

> Refer to steps to reproduce

**Would you like to work on the issue?**

@neel1998 , @ho-dor

neel1998, 2019-05-17:
I will look into it

mariobehling, 2019-06-01:
App upgraded. Issue not there anymore.

closed 2019-06-01 by mariobehling