fossasia/pslab-app #1759: App crashes on clicking export data option of multimeter

neel1998, 2019-05-31:
**Actual Behaviour**
App crashes on clicking export data option of multimeter

**Expected Behaviour**
App should not crash and a log file should be created

**Steps to reproduce it**
1. Record data in multimeter
2. Pause and select export data option

**LogCat for the issue**
2019-05-31 12:45:57.070 23742-23742/io.pslab E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.pslab, PID: 23742
    java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.io.File.exists()' on a null object reference
        at io.pslab.others.CSVLogger.writeCSVFile(CSVLogger.java:83)
        at io.pslab.activity.MultimeterActivity.onOptionsItemSelected(MultimeterActivity.java:428)
        at android.app.Activity.onMenuItemSelected(Activity.java:3469)
        at android.support.v4.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:436)
        at android.support.v7.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:196)
        at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
        at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
        at android.support.v7.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:64)
        at android.support.v7.widget.Toolbar$1.onMenuItemClick(Toolbar.java:204)
        at android.support.v7.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
        at android.support.v7.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:840)
        at android.support.v7.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
        at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:991)
        at android.support.v7.view.menu.MenuPopup.onItemClick(MenuPopup.java:128)
        at android.widget.AdapterView.performItemClick(AdapterView.java:318)
        at android.widget.AbsListView.performItemClick(AbsListView.java:1158)
        at android.widget.AbsListView$PerformClick.run(AbsListView.java:3153)
        at android.widget.AbsListView$3.run(AbsListView.java:4068)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6626)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:811)

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
yes

neel1998, 2019-05-31:
@CloudyPadmal  why do we have a pause button? Shouldn't it be made like other devices, where we have a stop button and clicking on it creates a csv file of logged data?

CloudyPadmal, 2019-05-31:
@neel1998 yes, it should be a stop button

neel1998, 2019-05-31:
@CloudyPadmal  i am thinking of inheriting multimeter from PSLab sensor class to avoid redundancy . Does it sound right?

CloudyPadmal, 2019-05-31:
That would be hard. As PSLab Sensor class got graphs and all which doesn't come with Multimeter. We can stick with the existing activity for Multimeter.

neel1998, 2019-05-31:
@CloudyPadmal  but if we want to show log data of multimeter in log data activity and playback that data.. That would become complex using current class, since realm data storing and playing back is already there in PSLab sensor , i think we can use it. This is what i think. Please give your suggestion

neel1998, 2019-05-31:
@CloudyPadmal  to resolve the #1373 I was thinking of recording data at certain time interval (say 500 ms) when record button is pressed. We will record the data of the element to which the multimeter knob is pointing .Does this sound okay ?

neel1998, 2019-05-31:
Also should i remove the show logged data option from multimeter menu since we are not displaying multimeter logs in datalogger activity

CloudyPadmal, 2019-05-31:
> @CloudyPadmal but if we want to show log data of multimeter in log data activity and playback that data.. That would become complex using current class, since realm data storing and playing back is [...]

If that's the case, best thing is to make a super class with all the graph and data storing functionalities and have `PSLabSensor` class inherit that.

CloudyPadmal, 2019-05-31:
> @CloudyPadmal to resolve the #1373 I was thinking of recording data at certain time interval (say 500 ms) when record button is pressed. We will record the data of the element to which the multimeter knob is pointing .Does this sound okay ?

Let's give user the option to set recording interval from configurations

neel1998, 2019-05-31:
> If that's the case, best thing is to make a super class with all the graph and data storing functionalities and have `PSLabSensor` class inherit that.

Do we actually need to show multimeter logs in data logger activity. I don't see any use of replaying multimeter data

neel1998, 2019-05-31:
> Let's give user the option to set recording interval from configurations

Should i open another issue regarding this or work with the already open one with continuous logging?

CloudyPadmal, 2019-05-31:
@neel1998 , let's see. May be we can plot the values in a graph in a timely manner? :thinking: That way we can replay.

neel1998, 2019-05-31:
> If that's the case, best thing is to make a super class with all the graph and data storing functionalities and have `PSLabSensor` class inherit that.

Please explain this a bit, because i think we can do this for all the instruments that inherit `PSLabSensor`

neel1998, 2019-05-31:
> @neel1998 , let's see. May be we can plot the values in a graph in a timely manner?  That way we can replay. And open a new issue for the configuration part and work on it..

Yes that can be done and would look good. But for that we need to store logs in realm and that would be easier if we had a parent class with all the methods

neel1998, 2019-05-31:
> Let's give user the option to set recording interval from configurations

What should be the range for the interval?

CloudyPadmal, 2019-05-31:
Multimeter extends GraphRealmClass 
PSLabSensor extends GraphRealmClass
LuxSensor extends PSLabSensor

Makes sense? Java.

neel1998, 2019-05-31:
@CloudyPadmal yes I will try implementing the GraphRealmClass. But I think I will fix #1762 with current structure and then raise a new issue to add new parent class. Will it be okay?

PR #1761 (2019-06-01): Fix #1759 multimeter record bug fixed
Fixes #1759 

**Changes**: Bug fixed, pause button replaced with stop button, csv file created on clicking stop button.

**Screenshot/s for the changes**:
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+ [... cut]
commit 9c2313be (2019-06-01): Fix #1759 multimeter record bug fixed (#1761)

* multimeter record bug fixed

* stop button icon changed
closed 2019-06-01 by CloudyPadmal