fossasia/pslab-app #1763: Add feature to preview the wave to be generated from the wave generator

neel1998, 2019-05-31:
**Actual Behaviour**
Currently user cant visually see the wave that will be generated unless it is opened in oscilloscope 

**Expected Behaviour**
Give user option to preview the wave in wave generator itself

**Would you like to work on the issue?**
Yes.

ho-dor, 2019-06-17:
@neel1998 this can be done without the board ? Should I work on this ?

neel1998, 2019-06-17:
@ho-dor you can. But I don't know whether we need this feature or not. @CloudyPadmal can you please confirm . And if we need it please add this issue to Milestone 4

CloudyPadmal, 2019-06-17:
Yes @neel1998 , Wave Generator UI is already pretty compact and it's a bit hard to figure out buttons too. Having a preview; I'm not sure if that's a good idea. We already giving user the shape of waveform (sine, triangular etc). Devices in market also don't provide such feature. So I'll close this for now.

closed 2019-06-17 by CloudyPadmal