fossasia/pslab-app #1815: App crashes when user plays a saved gyroscope log

neel1998, 2019-06-21:
**Actual Behaviour**
App crashes when user plays a saved gyroscope log

**Expected Behaviour**
App should not crash

**Steps to reproduce it**
Record gyroscope data, and play it from data logger activity

**LogCat for the issue**
2019-06-21 22:19:31.751 17761-17761/io.pslab E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.pslab, PID: 17761
    java.lang.NullPointerException: Attempt to invoke virtual method 'int java.util.ArrayList.size()' on a null object reference
        at io.pslab.fragment.GyroscopeDataFragment.playData(GyroscopeDataFragment.java:286)
        at io.pslab.models.PSLabSensor.onOptionsItemSelected(PSLabSensor.java:339)
        at android.app.Activity.onMenuItemSelected(Activity.java:3469)
        at android.support.v4.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:436)
        at android.support.v7.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:196)
        at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
        at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
        at android.support.v7.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:64)
        at android.support.v7.widget.Toolbar$1.onMenuItemClick(Toolbar.java:204)
        at android.support.v7.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
        at android.support.v7.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:840)
        at android.support.v7.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
        at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:991)
        at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:981)
        at android.support.v7.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
        at android.support.v7.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:151)
        at android.view.View.performClick(View.java:6297)
        at android.view.View$PerformClick.run(View.java:24797)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6626)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:811)

**Would you like to work on the issue?**
Yes

neel1998, 2019-06-22:
@CloudyPadmal  i think you closed this by mistake

CloudyPadmal, 2019-06-22:
Ah yes. #1820 PR title says it's fixing this issue.

neel1998, 2019-06-22:
Sorry. My bad .

CloudyPadmal, 2019-06-22:
No worries..

ho-dor, 2019-06-24:
I don't have the gyroscope sensor in my device so I don't think I will be able to debug it

neel1998, 2019-06-24:
No problem I will take it up

PR #1820 (2019-06-22): Fix #1815 Single Capture class created to handle all channels
Fixes #1816 Fixes #1806

**Changes**: Single Capture class created to handle all channels

**Screenshot/s for the changes**:
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L [... cut]
commit 413d03cb (2019-06-22): Fix #1815 Single Capture class created to handle all channels (#1820)

* Single Capture class created to handle all channels

* codacy and minor changes
PR #1831 (2019-06-26): Fix 1815 bug fixes in gyroscope
Fixes #1815 

**Changes**:bug fixed in gyroscope

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines or extra spaces [... cut]
closed 2019-06-26 by CloudyPadmal