fossasia/pslab-app #1817: Add Fourier Transform feature for CH3 and Built In MIC

neel1998, 2019-06-22:
**Actual Behaviour**
Currently Fourier Transform option is only for CH1 and CH2.

**Expected Behaviour**
Add Fourier Transform option for CH3 and builtin MIC

**Would you like to work on the issue?**
Yes

neel1998, 2019-06-25:
This one is not yet fixed entirely. Fourier Transform for Built In mic input is still to be implemented. Reopening this issue

CloudyPadmal, 2019-06-25:
Don't add tag lines like "Fix #nnn" in PR body. They will automatically closes issues. Also I got a question. I guess the solution for this issue is something quiet similar to what you've done in #1799 addressing #1798. And this issue is opened for 3 days. Can we get a quick fix for this?

neel1998, 2019-06-25:
No actually after converting all capture, audio buffer classes to one, it will be different to have fourier for Built in Mic Audio Input. Still i will make a PR by tonight

PR #1826 (2019-06-25): Fix #1821 converted audiobuffer, fft and channel capture classes to single capture class

Fixes #1821 

**Changes**: Everything is now handled through single Capture Class

**Screenshot/s for the changes**:
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+< [... cut]
PR #1830 (2019-06-25): Fix #1817 fourier transform for built in mic input working fine
Fixes #1817 

**Changes**: Fourier Transform added for built in mic

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My co [... cut]
commit 5a50323e (2019-06-25): Merge pull request #1830 from neel1998/fft_mic

Fix #1817 fourier transform for built in mic input working fine
closed 2019-06-25 by cweitat