fossasia/pslab-app #1840: Expand sensors support

cweitat, 2019-06-29:
To expand the support of sensors
https://pslab.io/sensors/

Sensors from the kit is the priority followed by commonly used sensors.

- [ ] Moisture sensor FR-04
- [ ] LDR GL5528
- [ ] Microphone KY-037
- [ ] Magnetic KY-003
- [ ] Temperature LM-35
- [ ] Gas MQ135

ho-dor, 2019-06-29:
I'll start on this 👍 .

neel1998, 2019-06-30:
I am working on MQ135 Gas sensor

neel1998, 2019-06-30:
@cweitat most of the mentioned sensors are to I2C as i understand. So, how do we communicate to sensors which are not I2C that are connected to the PSLab device. Please guide

woshikie, 2019-06-30:
@neel1998
Googling the part number with arduino could be a good start. 
Some sensors are more straight forward as they are either resistive or capacitive, while some might use other serial communication line such as UART.

neel1998, 2019-06-30:
@woshikie thanks for the help. I will look into it