fossasia/pslab-app #1860: Robotic Arm time line control buttons not visible

neel1998, 2019-07-11:
**Actual Behaviour**
Currently play/pause/stop buttons are not visible

**Expected Behaviour**
Buttons should be visible

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
yes

ho-dor, 2019-07-11:
[image]

Visible on my device.

neel1998, 2019-07-11:
@ho-dor  may be the size of the layouts are assigned properly from java, so it is looking different on different screen size. I will look into it

PR #1871 (2019-07-14): robotic arm control panel now visible
Fixes #1860 

**Changes**: robotic arm control panel now visible

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra [... cut]
closed 2019-07-14 by CloudyPadmal