fossasia/pslab-app #1862: Auto Start App option not working

neel1998, 2019-07-11:
**Actual Behaviour**
Even if the auto start option in the app settings in not selected, app gets open when the board is connected

**Expected Behaviour**
App should only open if the auto start app option is selected

**Steps to reproduce it**
1. Uncheck the Auto Start Option from settings menu
2. Connect the PSLab board

**Would you like to work on the issue?**
Yes

CloudyPadmal, 2019-08-11:
Why is this closed without providing any comment? :thinking:

neel1998, 2019-08-11:
@CloudyPadmal sorry . I forgot to put the comment. Basically the usb attach listener is added using intent filter... And according to this https://stackoverflow.com/questions/4862607/how-to-add-or-remove-intent-filter-programmatically-in-android we can't programmatically add/remove intent-filter. There might be some lousy workaround, that too maybe. Since this isn't a crucial change. We can close this for now.

closed 2019-08-11 by CloudyPadmal