fossasia/pslab-app #1864: Twitter and Facebook links not working in About screen

neel1998, 2019-07-11:
**Actual Behaviour**
Twitter and Facebook links not working in About screen

**Expected Behaviour**
Links should direct to respective pages

**Steps to reproduce it**
Click on facebook/twitter links in About page

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

neel1998, 2019-07-11:
@mariobehling  can you please provide the links for the facebook and twitter accounts. Thanks

CloudyPadmal, 2019-07-11:
@neel1998 haven't you followed the twitter account and fb page yet?

neel1998, 2019-07-11:
fb : https://www.facebook.com/fossasia/?ref=br_rs
twitter : https://twitter.com/fossasia
Are these two links correct?

CloudyPadmal, 2019-07-11:
Search PSLab and you'll see

CloudyPadmal, 2019-07-11:
Twitter link is already in the readme. Always need guidance.

neel1998, 2019-07-11:
Got it. Sorry for bothering you.

PR #1867 (2019-07-12): about facebook/ twitter page links changed
Fixes #1864 

**Changes**:Correct links added

**Screenshot/s for the changes**:
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines or extra [... cut]
closed 2019-07-12 by CloudyPadmal