fossasia/pslab-app #1880: Reduce repo size

CloudyPadmal, 2019-07-19:
**Actual Behaviour**

pslab-android repo size is 1.9 GB. https://api.github.com/repos/fossasia/pslab-android

**Expected Behaviour**

Need to reduce it to actual size by removing unwanted cached files similar to what they've done in phimpme project

**Steps to reproduce it**

https://api.github.com/repos/fossasia/pslab-android check size property. Or try to clone the repo

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Issue for GSoC

neel1998, 2019-07-19:
on it :+1:

neel1998, 2019-07-28:
@CloudyPadmal  I have found the reason for huge repo size. There are lots of old apks with size around 20 Mb. You can see those files by executing the following command,

``` git rev-list --objects --all | git cat-file --batch-check='%(objecttype) %(objectname) %(objectsize) %(rest)' | sed -n 's/^blob //p' | sort --numeric-sort --key=2 | cut -c 1-12,41- | numfmt --field=2 --to=iec-i --suffix=B --padding=7 --round=nearest ```

As can be seen the apks are causing the repo size to explode. I have checked the bfg repo cleaner and we can use it to remove these unnecessary apks. You can download the bfg.jar from this link https://rtyley.github.io/bfg-repo-cleaner/#download and then execute the following command, 

``` java -jar bfg.jar --delete-files *.apk ```

This will clean all the apks. After that use the command below,

``` git reflog expire --expire=now --all && git gc --prune=now --aggressive ```

and then force push. Since you have the push access, i guess you will have to do this. Let me know if i am making some mistake. Thanks

CloudyPadmal, 2019-07-29:
Yes I can do this. But before that, run all these commands in a fresh clone and document the following (as much details as possible). Probably in a text file and upload here.
1. git log ($ git log)
2. file structure ($ tree .)
3. command outputs on console from pruning and bfg tool

neel1998, 2019-07-29:
Sure

neel1998, 2019-07-29:
[repo_clean.zip](https://github.com/fossasia/pslab-android/files/3442840/repo_clean.zip)

@CloudyPadmal  please find all the requested results in the above zip file in .txt formats. The file names are self-explanatory. Please let me know if anything else has to be done. Thanks

CloudyPadmal, 2019-07-30:
@neel1998 you're in PSLab-Admin group. You'll be able to make a push to main repo. https://github.com/orgs/fossasia/teams/pslab-admins Can you proceed with this?

CloudyPadmal, 2019-07-30:
Along with this, update the scripts to delete the old apks. 

https://github.com/fossasia/open-event-organizer-android/blob/development/scripts/update-apk.sh#L56

Otherwise this issue will happen again.

adityastic, 2019-07-30:
@CloudyPadmal Sir, @neel1998 has added the script to delete apk's, I remember reviewing them as well.

adityastic, 2019-07-30:
Yes, I can confirm he did. 

https://github.com/fossasia/pslab-android/blob/development/scripts/upload-apk.sh#L15

neel1998, 2019-07-30:
@CloudyPadmal  i think you only will have to make these changes since the branch is protected. When i tried force pushing i got the following error,

``` 
git push -f upstream 
warning: push.default is unset; its implicit value has changed in
Git 2.0 from 'matching' to 'simple'. To squelch this message
and maintain the traditional behavior, use:

  git config --global push.default matching

To squelch this message and adopt the new behavior now, use:

  git config --global push.default simple

When push.default is set to 'matching', git will push local branches
to the remote branches that already exist with the same name.

Since Git 2.0, Git defaults to the more conservative 'simple'
behavior, which only pushes the current branch to the corresponding
remote branch that 'git pull' uses to update the current branch.

See 'git help config' and search for 'push.default' for further information.
(the 'simple' mode was introduced in Git 1.7.11. Use the similar mode
'current' instead of 'simple' if you sometimes use older versions of Git)

Username for 'https://github.com': neel1998
Password for 'https://neel1998@github.com': 
Counting objects: 12625, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (2818/2818), done.
Writing objects: 100% (12625/12625), 8.91 MiB | 4.52 MiB/s, done.
Total 12625 (delta 8018), reused 12245 (delta 7666)
remote: Resolving deltas: 100% (8018/8018), done.
remote: error: GH006: Protected branch update failed for refs/heads/development.
remote: error: Cannot force-push to a protected branch
To https://github.com/fossasia/pslab-android.git
 ! [remote rejected] development -> development (protected branch hook declined)
error: failed to push some refs to 'https://github.com/fossasia/pslab-android.git'
```
Also please remove png files from the history along with apks with following command,
``` java -jar bfg.jar --delete-files *.png ```

CloudyPadmal, 2019-07-30:
I've made a backup branch and removed protection for now. Try a force push now.

neel1998, 2019-07-31:
@CloudyPadmal we can close this now.

closed 2019-07-31 by CloudyPadmal