fossasia/pslab-app #1882: PSLab SD Card configuration

CloudyPadmal, 2019-07-20:
**Actual Behaviour**

In order to use PSLab without using a device (say, mount on a weather station), a configuration file is needed to save in SD card so that the PSLab can read the setting from this file and do the processing and logging

**Expected Behaviour**

Have these properties in the config file in YAML format
- File Name (Optional)
- Instrument Name
- Logging Interval
- Parameters needed to be log (Depending on the instrument, Oscilloscope:: CH1, CH2 etc..)

**Steps to reproduce it**

> Not available

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Issue for GSoC

neel1998, 2019-07-21:
@CloudyPadmal user should be able to give start time and end time , right? To specify the period in which the data should be logged for given parameters.

CloudyPadmal, 2019-07-21:
No. It will make things complicated.  Just add the configs mentioned in the issue body.

neel1998, 2019-07-21:
What should be the unit of logging interval, minutes / hours / days? Should there be an option for that as well?

CloudyPadmal, 2019-07-22:
Yea put an option

PR #1905 (2019-08-01): create config file working fine
Fixes #1882 
**Changes**: User can create config files

**Screenshot/s for the changes**:

[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines or extra  [... cut]
closed 2019-08-01 by cweitat