fossasia/pslab-app #1901: Master branch Playstore APKs not being built

CloudyPadmal, 2019-07-31:
**Actual Behaviour**

Once a release is made (i.e. push to master), master apks should be generated and supply them to fastlane for Google Play store. Build is ending without proceeding to next steps from https://github.com/fossasia/pslab-android/blob/master/scripts/upload-apk.sh#L24

**Expected Behaviour**

Generate signed apks and fastlane sends them to Google Play

**Steps to reproduce it**

> Not available

**LogCat for the issue**

https://travis-ci.org/fossasia/pslab-android/builds/565944930#L1705

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

@neel1998

adityastic, 2019-07-31:
Directory needs to be changed, let me drop a pr in a minute

neel1998, 2019-07-31:
@adityastic yes we are in apk directory. We need to use ../gradlew

adityastic, 2019-07-31:
@neel1998 no xD that wont work. ( cd .. & ./gradlew ... ) is what you need to do.

adityastic, 2019-07-31:
but for the case of debugging, let us store the aab for debug as well.

neel1998, 2019-07-31:
@adityastic got it.

CloudyPadmal, 2019-07-31:
Didn't get fixed

CloudyPadmal, 2019-07-31:
Refer this https://github.com/fossasia/open-event-attendee-android/blob/development/scripts/update-apk.sh

neel1998, 2019-07-31:
@adityastic i guess we are not giving the correct path for the aab.... @CloudyPadmal the problem is only occurring when trying to upload using fastlane, right?

CloudyPadmal, 2019-07-31:
It doesn't even reach that far. Basic file handling errors

neel1998, 2019-07-31:
@CloudyPadmal done. Please check. Hopefully, it should work as expected this time

CloudyPadmal, 2019-08-01:
https://travis-ci.org/fossasia/pslab-android/builds/566339963#L1748

Still errors. I'll look into this issue

CloudyPadmal, 2019-08-02:
Fixed

PR #1902 (2019-07-31): chore: updated build script to generate bundles
Fixes #1901 

Changes:
- Updated Script to support aabs for both branches
PR #1904 (2019-08-01): upload script error fixed
Fixes #1901
closed 2019-08-02 by CloudyPadmal