fossasia/pslab-app #1925: Add documentation for ESP communication and connection

neel1998, 2019-08-12:
**Actual Behaviour**
Currently no docs available regarding ESP connection and communication

**Expected Behaviour**
Add docs with following details
1. How to attach ESP chip
2. How to set up the chip
3. How to communicate with the PSLab over Wifi

**Steps to reproduce it**
N/A

**LogCat for the issue**
N/A

**Screenshots of the issue**
N/A

**Would you like to work on the issue?**
Yes

DARK-art108, 2020-05-07:
Can I work on this issue?

adityastic, 2020-05-07:
> Can I work on this issue?

Go ahead

DARK-art108, 2020-05-08:
The above docs should be added in Pocket science Lab documentation Repo under tutorials section?

hariom9522, 2020-11-28:
can i work on this issue

limitless-Sky, 2020-11-30:
@neel1998 can I work on this issues?

AtigPurohit, 2020-12-07:
can i work on this issue?

marcnause, 2024-11-09:
I think this ticket refers to the PSLab v5 board which is superseded by the v6 board now. ESP8266 support was never included in the firmware as far as  know. 

Support for the ESP-01 and the v6 board is developed in https://github.com/fossasia/pslab-android/issues/2509.

Can this issue be closed?

closed 2024-11-09 by mariobehling