fossasia/pslab-app #1944: Google Play warnings

CloudyPadmal, 2019-08-16:
**Actual Behaviour**

Several warnings were given by Google Play related to APIs we are using.
> The following APIs are greylisted and Google can’t guarantee that they will work on existing versions of Android. Some may be already be restricted for your target SDK

**Expected Behaviour**

Need to address them before they become unstable

**Steps to reproduce it**

> Pre-launch report

**LogCat for the issue**

```
StrictMode policy violation: android.os.strictmode.NonSdkApiUsedViolation: Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V
	at android.os.StrictMode.lambda$static$1(StrictMode.java:428)
	at android.os.-$$Lambda$StrictMode$lu9ekkHJ2HMz0jd3F8K8MnhenxQ.accept(Unknown Source:2)
	at java.lang.Class.getDeclaredMethodInternal(Native Method)
	at java.lang.Class.getPublicMethodRecursive(Class.java:2075)
	at java.lang.Class.getMethod(Class.java:2063)
	at java.lang.Class.getMethod(Class.java:1690)
	at android.support.v7.widget.ViewUtils.makeOptionalFitsSystemWindows(ViewUtils.java:84)
	at android.support.v7.app.AppCompatDelegateImpl.createSubDecor(AppCompatDelegateImpl.java:685)
	at android.support.v7.app.AppCompatDelegateImpl.ensureSubDecor(AppCompatDelegateImpl.java:518)
	at android.support.v7.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:466)
	at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:140)
	at io.pslab.activity.MainActivity.onCreate(MainActivity.java:109)
	at android.app.Activity.performCreate(Activity.java:7144)
	at android.app.Activity.performCreate(Activity.java:7135)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1271)
	at androidx.test.runner.MonitoringInstrumentation.callActivityOnCreate(MonitoringInstrumentation.java:184)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2931)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3086)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:78)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:108)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:68)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1816)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6718)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
```

```
StrictMode policy violation: android.os.strictmode.NonSdkApiUsedViolation: Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z
	at android.os.StrictMode.lambda$static$1(StrictMode.java:428)
	at android.os.-$$Lambda$StrictMode$lu9ekkHJ2HMz0jd3F8K8MnhenxQ.accept(Unknown Source:2)
	at java.lang.Class.getDeclaredMethodInternal(Native Method)
	at java.lang.Class.getMethod(Class.java:2064)
	at java.lang.Class.getDeclaredMethod(Class.java:2047)
	at android.support.v7.widget.ViewUtils.<clinit>(ViewUtils.java:44)
	at android.support.v7.widget.ViewUtils.makeOptionalFitsSystemWindows(ViewUtils.java:80)
	at android.support.v7.app.AppCompatDelegateImpl.createSubDecor(AppCompatDelegateImpl.java:685)
	at android.support.v7.app.AppCompatDelegateImpl.ensureSubDecor(AppCompatDelegateImpl.java:518)
	at android.support.v7.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:466)
	at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:140)
	at io.pslab.activity.SplashActivity.onCreate(SplashActivity.java:29)
	at android.app.Activity.performCreate(Activity.java:7144)
	at android.app.Activity.performCreate(Activity.java:7135)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1271)
	at androidx.test.runner.MonitoringInstrumentation.callActivityOnCreate(MonitoringInstrumentation.java:184)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2931)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3086)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:78)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:108)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:68)
[... 89 lines cut ...]
	at okhttp3.internal.platform.AndroidPlatform$CloseGuard$Companion.get(AndroidPlatform.kt:306)
	at okhttp3.internal.platform.AndroidPlatform.<init>(AndroidPlatform.kt:47)
	at okhttp3.internal.platform.AndroidPlatform$Companion.buildIfSupported(AndroidPlatform.kt:365)
	at okhttp3.internal.platform.Platform$Companion.findPlatform(Platform.kt:204)
	at okhttp3.internal.platform.Platform$Companion.access$findPlatform(Platform.kt:178)
	at okhttp3.internal.platform.Platform.<clinit>(Platform.kt:179)
	at okhttp3.OkHttpClient.<init>(OkHttpClient.kt:211)
	at okhttp3.OkHttpClient.<init>(OkHttpClient.kt:204)
	at io.pslab.fragment.ESPFragment$ESPTask.doInBackground(ESPFragment.java:76)
	at io.pslab.fragment.ESPFragment$ESPTask.doInBackground(ESPFragment.java:64)
	at android.os.AsyncTask$2.call(AsyncTask.java:333)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:764)
```

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Issue for GSoC

tusharr-98, 2019-10-08:
@neel1998 @CloudyPadmal  can you fill me in about the expected behavior so that I can give it a try

anurag0singh, 2021-01-05:
@CloudyPadmal I can try to make a pre-launch report by using the google play console and android studio and by making a Robo script. Can I work on this?

closed 2026-02-08 by marcnause